robot-army/hotstick #12: Research other similar projects

robot-army, 2015-10-14:
There are many other projects of this sort. Let's make sure we pick the good bits from them and adhere to their licensing conditions. 

Let's either capture these good bits as requirements or methods

robot-army, 2015-10-14:
https://github.com/albert21martins/SolderingStation-master
https://www.youtube.com/watch?v=uHT_JV5KMJI

http://hackaday.com/2015/05/11/hacking-amazing-soldering-features-into-the-already-great-weller-wmrp/
https://github.com/FlyGlas/WMRP

https://debugginglab.wordpress.com/2014/10/30/soldering-station/

https://github.com/ConnyCola/SolderingStation

As a start...

jeremypoulter, 2016-02-02:
This thread also contains some useful info http://dangerousprototypes.com/2012/02/27/weller-rt-type-tip-teardown-and-reverse-engineering/

jeremypoulter, 2016-02-06:
Here is another general purpose soldering controller; http://dangerousprototypes.com/forum/viewtopic.php?f=56&t=7218&p=61175#p61175

jeremypoulter, 2016-04-06:
Just spotted another project to do a Weller RT controller, this one however is trying to get everything in the pencil. Think I like our solution better but could be a good source for reference;

https://hackaday.io/project/10140-soldering-pen

jeremypoulter, 2016-04-23:
Another generic soldering station,this time runing at 110v https://pcbsmoke.wordpress.com/2016/04/21/110v-temperature-controlled-soldering-station-build-no-2/