robot-army/hotstick #15: Display fancy graphics in colour

robot-army, 2015-10-14:


Gavatastic, 2015-10-14:
compatible with a range of displays (to avoid obsolescence if screen component no longer available) [if Arduino based, something like Adafruit's GFX library?]

daubers, 2015-10-14:
How about http://www.adafruit.com/products/358

robot-army, 2015-10-14:
Should we steer away from specific details of implementation for the
moment? What's good about that specific part? Let's capture those good bits
in requirements, and figure out how to implement them in the design stage
of the project yeah?

Remulos, 2015-10-14:
In that case one of the requirements for the display, as well as the rest of the device, needs to be the simplicity of it's programming. If the main target market is hobbyists and hackspaces then hack-ability is a requirements and some obsucre display which requires an equally obscure driver isn't ideal. Something like the Library is accesable to all and already comes with loads of online reseources.

robot-army, 2015-10-14:
Yep, happy with that additional requirement. I'm sure whatever we end up
choosing won't be esoteric... we do want to be able to make this work,
after all!

closed 2015-11-05 by robot-army