robot-army/hotstick #16: BOM Cost and time to build makes it worth it

robot-army, 2015-10-14:
The BOM cost plus the time to build it (at eg.£10/hour), must be cheaper than a WSM 1C, otherwise we might as well just buy one of them.

Approx £330. 

http://uk.farnell.com/weller/wsm-1c-uk-eu/soldering-station-40w-230v-battery/dp/1729966

dhprice, 2015-10-14:
Different users will likely have different cost sensitivities. Perhaps have three existing products in each usage category as the one to beat on performance and price.

robot-army, 2015-10-14:
I think we should focus on making one model only.

We can reject certain requirements (eg. battery power) in order to satisfy
the lower cost requirements. That's the only one that I could see us
rejecting for cost sensitivity at the moment...(unless we want to do it
some other way than by using Weller tips, which I wouldn't be stoked about
to be honest, they're REALLY good)

Remulos, 2015-10-14:
When looking at components I think that we can save money by avoiding the top spec ones with high degrees of accuracy. Realistically, most applications won't need the iron to be heated to within 5 degrees of the set temperature, more likely to within 10 or 25. We should prioritise ruggedness rather than accuracy.

robot-army, 2015-10-14:
Component choices are going to save us a small percentage rather than tens
or hundreds of pounds overall.

The requirements will set most of the cost. Think battery, colour LCD,
optical encoder for setpoint adjustment, IP/shock and vibration rating, CE
compliance, flammability requirements for materials, that sort of thing is
where the major design decisions will come from.

The idea behind writing this requirement was that if we can't meet this
one, we might as well not do the project.

dhprice, 2015-10-15:
But why would a hobbyist pay more than £100? If you design a modular architecture then you could satisfy the pro with a £300 unit and the hobbyist with a £100 unit while retaining a the majority of common features.

robot-army, 2015-10-15:
I paid just about £100 for the first iron I bought as a hobbyist:
http://uk.farnell.com/ersa/i-con-pico/soldering-station-80w-240v-eu/dp/2057377

It's very good, but if we build one with all the listed requirements so far
for even £50 more, it would be by far better value.

dhprice, 2015-10-15:
I quite like the look of the pico packaging..

My point is that £150 might be far better value to you but not to someone else so perhaps the approach should be for a design, a platform, that allows people to customise elements that address their prefered values.

robot-army, 2015-10-15:
It's good. Key requirements here that it's missing are the quick-change
tips and the sensible idle modes. The pico watches power output, and if it
sees that you haven't made a joint for x minutes it goes to 250 deg. Then,
it keeps sensing, if it dissipates at 250 that means its had the tip wiped
in the scrubber or seen a light breeze and it quickly comes up to temp.

If it sits at 250 for xx minutes, it shuts down to 60 deg and needs a
button pushed to come back on.

I think the adaptability will be provided for by the open-source nature of
the design. To be honest, it would best capture my interest to design one
product that meets all the requirements we accept, and in future adjust the
requirements for other versions (rather than try to make a 'master of all
trades' first).

To be clear, we don't need to accept all the requirements here! They're
just ideas at this stage and have cost nothing but time so far.

dhprice, 2015-11-04:
Accepted

closed 2015-11-05 by robot-army