robot-army/hotstick #17: Single or dual station?

robot-army, 2015-10-14:
Do we want this thing to be able to drive two irons? (one big, one small, or held as chop-sticks for removing SMD passives?)

jeremypoulter, 2015-10-14:
Could dual outputs also be used for a hot air gun?

robot-army, 2015-10-14:
I'm not sure that there's a hot air gun that can be easily just 'fed volts'
to make it go (equivalent to the Weller tips). If there is, integrating it
would be awesome, but for now I think it'd be more realistic to stick to
multiples of the weller tips.

One thing that could be quite interesting would be to run both outputs into
two tips held in some sort of hot-tweezer arrangement.

plett, 2015-10-14:
If the hotstick controller is cheap enough, I'd just make a second one if I wanted to control a second iron, so don't see a need for that in the design. Hot-tweezers might be interesting, but I don't know enough about the design of the tweezers to know if they need dual-channel control.

robot-army, 2015-10-14:
OK, good point. Single-station for now.

closed 2015-10-14 by robot-army