robot-army/hotstick #18: Design for compliance

robot-army, 2015-10-14:
The design work should be undertaken with IEC 60335 (Household and similar electrical appliances) standards in mind.

robot-army, 2015-10-14:
Let's start with a review of the standard and make sure that a) this is the right one and b) we've got a hope in hell of meeting it!!

Remulos, 2015-10-15:
The specific standard would be IEC 60335-2-45 "Household and similar electrical appliances – Safety – Part 2-45: Particular requirements for portable heating tools and similar appliances".

I can provide a copy of the standard if we keep it confidential, i.e. no publishing it on open access sites like the google group/ wiki/ Github

Remulos, 2015-10-15:
Here is the full IEC 60335-1 document, I don't think that it includes the latest amendment but that's to do with microwave ovens or something so not relevant. 
http://www.anbotek.com.cn/standard/(%E5%AE%B6%E7%94%B5)Household-EN%2060335-1_2012.pdf

I've taken a look at Part 2-45 and made a note of all the differences relevant to a soldering iron and attached here. This is for a Soldering Iron and not a Soldering Gun or an Induction Soldering iron.

With regards to compliance as a whole, the requirement for this product really just depends on the end goal. It's always a good idea to design for compliance but if we're not selling it then we don't need to actually get it tested. If we're selling it by mail order to 20 people a year then we don't need to test it. If we're going to sell it on the Sparkfun website or whatever then we will need to go through the whole compliance process. You don't need to get any sort of certification for the schematic if we publish it online and can't be held liable for anything if we publish it and it hurts someone who has built it. We can be held liable if we manufacturer it and sell it as a kit, regardless of whether it's open-source or not.
[IEC 60335-2-45 Gap Analysis.pdf](https://github.com/robot-army/hotstick/files/14050/IEC.60335-2-45.Gap.Analysis.pdf)

robot-army, 2015-10-15:
Excellent! Nice work!

So, I guess the key thing is to extract all the relevant requirements from
these document and turn them into bite-sized manageable requirements?

Remulos, 2015-10-15:
Yer the standard is about 200 pages long but a lot of it is just about how the tests should be carried out. I am looking now and will see how long it'll take me to put together a concise requirements list.

robot-army, 2015-10-15:
That's a seriously arduous task I just suggested. This document is weighty.

Key question is all this jazz about supply cords. What do we gain by using
an external 'brick'? Is that a type X, Y or Z attachment?? What class are
we in? 0, 0l?

robot-army, 2015-10-15:
Where do I find 60335-2-45? It seems we've only got -1

Remulos, 2015-10-15:
I couldn't download it so I just took out all the bits pertinent to this project and put them in the pdf I attached. That pdf is all that is needed beyond  60335-1 unless we want to add a heat gun or the soldering iron is going to contain a transformer.

robot-army, 2015-10-15:
OK cool!

robot-army, 2015-11-05:
This is not a requirement, but it will spill out into other requirements. Looking forward to seeing what @remulos gets done