robot-army/hotstick #19: Define the target users

dhprice, 2015-10-14:
One way to categorise and qualify requirements is by target user/usage;

Hobbyist - usability and simplicity
Hackspace/Makerspace - ruggedness and safety
Professional - ?

robot-army, 2015-10-14:
Yes, this, target users will define what standards we need to comply with as well. 

To narrow it down, I doubt my workplace would buy an open-source iron, so let's aim high and go for the hackspace user. I think the durability required by the hackspace would make it an attractive option for a home user/hobbyist as well.

dhprice, 2015-10-15:
At this stage I would look to tag requirement and design choices with the usage class so the choice can be left to the builder. There may be requirements that would suit a hackspace but not be cost justifiable for a hobbyist, for example. If the first step in the analysis of requirements is to categorise them then the design rationale becomes easier.