robot-army/hotstick #21: Auxiliary software controlled switch

jeremypoulter, 2015-10-14:
We could add an auxiliary output/switch to control external equipment to turn on when the iron is on  and/or out of the holder. 

This could be used to switch on lighting, extraction or fan.

Could be as simple as a relay.

robot-army, 2015-10-14:
Oooo yes some opto-isolated GPIO's would be very handy!

plett, 2015-10-14:
Do we need to design that in to the board and increase the BOM, or just expose an I2C/SPI/whatever header and let people design their own relay board and patch the software to drive it?

robot-army, 2015-10-14:
I don't think there's a requirement for relays, but I think there's a
strong requirement for all the IO to be robust against ESD, overvoltage,
short circuit etc, and no requirement for it to be particularly fast or low
power.

robot-army, 2015-10-14:
What I mean by that, is that any IO on board should be optoisolated.

jeremypoulter, 2015-10-14:
Essentially my use case for this was that I always forget to turn on my fan when soldering it would be really handy if the soldering station automatically switched this on, but generally any unused GPIO (suitably isolated) where available that would be sufficient. 

IC2 and SPI I think would be god, but IMHO I would just have these on an internal header rather than externally accessible as these probably would be only useful for custom expansions. The GPIO could have some built in 'if this then that' controls in the generic firmware, eg when iron removed from folder trun on aux 1.

Remulos, 2015-10-15:
I agree that this would be a really useful feature and does help to differentiate it from other irons but maybe this is a low priority requirement? Review a preliminary design once we have one and see if it's possible rather than making it an absolute requirement.

I also agree that the I2C and SPI should be on an internal header, that way we could quite easily allow for a number of different addresses using jumpers.

dhprice, 2015-11-04:
Accepted, it'll be nice to have some external device control broken out on headers on the PCB. Need to think about this for compliance.

closed 2015-11-05 by robot-army