robot-army/hotstick #4: Decide on a software license

robot-army, 2015-10-14:


jeremypoulter, 2015-10-14:
I normally use the Acache license or BSD license. The latter (BSD) is a nice simple which I quite like. The Apache license is also not bad and has some requirements about granting licenses to use any IPR that are not in other OSS licenses.

I would avoid GPL on principle, as it is totally hypocritical (especially v3). How can you preach that your license promotes choice while removing peoples choice at the same time... ?

plett, 2015-10-14:
It seems sensible to me for the software licence to have similar clauses to the hardware licence so the two parts can be used together by any potential manufacturers.

As per the other issue, we've picked CERN OHL 1.2 for the hardware licence. According to  http://www.ohwr.org/projects/cernohl/wiki "CERN–OHL is to hardware what the General Public Licence (GPL) is to software".

IANAL, but reading through http://www.ohwr.org/attachments/2388/cern_ohl_v_1_2.txt seems to say that it allows people to manufacture devices based on the hardware but that they must publish the design of the hardware they make. The closest software licence that I know of that meets the same clause is GPL. For personal reasons, I'd prefer GPLv2 over GPLv3.

robot-army, 2015-11-05:
I think one important goal with all the licensing stuff on here is to avoid the **** that happened with Arduino. I hope we can all learn from that mess.