robot-army/hotstick #6: Drive a Weller RP1 tip

robot-army, 2015-10-14:


robot-army, 2015-10-14:
Need more detail here: Current, voltage, connector and other physical interface details

daubers, 2015-10-14:
I can't actually find anything about an RP1 tip???

ribbotson, 2015-10-14:
Shouldn't this be an RT1 tip ?

robot-army, 2015-10-14:
Must be a typo! Sorry!

http://www.apexhandtools.com/weller/PDFs/Weller%20WX%20brochure%20US.pdf
lists all the tips available for WXMP irons.

plett, 2015-10-14:
http://dangerousprototypes.com/2012/02/27/weller-rt-type-tip-teardown-and-reverse-engineering/ (and http://dangerousprototypes.com/forum/viewtopic.php?f=19&t=3583 ) documents a teardown on a RT1 tip.

It apparently uses a 3-pin stereo headphone jack (unclear if 2.5mm or 3.5mm) for +v, GND and temp sense, with a k-type thermocouple which you read when power isn't being applied.

robot-army, 2015-11-04:
The system needs to be able to drive these tips at the maximum power.

closed 2015-11-05 by robot-army