robot-army/hotstick #25: Decide on Documentation licence

dhprice, 2015-10-15:
This should be at least as permissive as the software and hardware licences.

I suggest Creative Commmons  Attribution-ShareAlike International -  CC BY-SA 4.0

"This license lets others remix, tweak, and build upon your work even for commercial purposes, as long as they credit you and license their new creations under the identical terms. This license is often compared to “copyleft” free and open source software licenses. All new works based on yours will carry the same license, so any derivatives will also allow commercial use. This is the license used by Wikipedia, and is recommended for materials that would benefit from incorporating content from Wikipedia and similarly licensed projects."

http://creativecommons.org/licenses/by-sa/4.0/

plett, 2015-10-16:
IANAL, but CC BY-SA seems like a good match to the CERN OHL which has already been chosen for the hardware licence. Sounds good to me.

dhprice, 2015-11-04:
Yeah, do this.

dhprice, 2015-11-04:
Or don't? Who cares?

robot-army, 2015-11-05:
I think one important goal with all the licensing stuff on here is to avoid the **** that happened with Arduino. I hope we can all learn from that mess.