robot-army/hotstick #26: Ergonomics requirements

dhprice, 2015-10-15:
A key requirement and an umbrella category.

From a discussion with Richard I) we should address the ergonomics requirements of the controller and especially the handset.

How do we capture and define the requirements for the handset, meaning the tip, handle and cable? These characteristics will determin how "nice"/comfortable the unit is to use over a period.

For example, the cable on my Antex is very irritating requiring me to spend too much effort to arrange and manage while soldering. This bears on the cable material (slicone ?), thickness, weight and length.

robot-army, 2015-10-15:
Great, yep. This one is a tricky thing to capture. I think we should
require the cable to be very flexible and heatproof, and we should have the
base unit suitably heavy and sturdy, maybe that's a good start?

Remulos, 2015-10-15:
This a requirement that could really benefit from a sort of show and tell of the soldering irons we have. It's hard to quantify ergonomics in a way that makes sense to everyone so I would suggest we try to gather an array of irons and physically try each of them out one night.

robot-army, 2015-10-15:
Good idea, I'll bring my i-con pico. Maybe Bob Dunlop can bring one of the
wellers from my old work, and I think there's a Hakko at the space already.

Maybe we can make this coincide with the requirements review, and do that
in person/via webcast realtime.

Remulos, 2015-10-15:
I'll bring my £40 Maplin iron as an example of what NOT to do.

jeremypoulter, 2015-10-15:
I have a Hakko I can bring in.

dhprice, 2015-11-04:
Duplicate as near as dammit the weller handset and cable in terms of dimensions, weight, flexibility of cable.

dhprice, 2015-11-04:
Ergonomic requirements for the basestation is the rotary encoder with detents to change temperature etc quickly.

robot-army, 2015-11-05:
Captured in other requirements (handle mass, cable flexibility and rotary encoder). Duplicate.

closed 2015-11-05 by robot-army