robot-army/hotstick #27: (Ab)usage statistics

Gavatastic, 2015-10-15:
Collect data on how it is used (temps/duration) to better understand why rLab solder stations have limited life

robot-army, 2015-10-15:
Totally, this is part of measuring and recording current.

Remulos, 2015-10-15:
What sort of range would you want to store? Full lifecycle stats (i.e. 10,000 uses or something) might exceed basic micro controller memory.

dhprice, 2015-10-15:
£2.50 or so for an ESP8266 WiFi serial interface ... you could store the 
data anywhere you like.

Remulos, 2015-10-15:
If we're putting wireless in this thing then we'll have to look at some extra compliance requirements so we should figure this one out soon.

robot-army, 2015-10-15:
Oh man, intentional radiators :)

I'd much rather an SD card or something

robot-army, 2015-10-15:
but again, implementation is not for now. If we want to store stuff, we can
make it a requirement and figure out how to implement it 'soon'.

Gavatastic, 2015-10-15:
Need not take to much memory if we're only storing and adjusting aggregate stats. The key thing would be to decide which one - no of usages, duty rate, length of low power periods ie are people just lazily leaving it on etc (not an expert on what would be best)

I like the idea of transmitting them for storage elsewhere, but think it's probably a lower priority requirement?

robot-army, 2015-10-15:
I really don't like the idea of wireless transmissions if we're going to
try to get this thing past any real compliance or type approval testing.

I'm not saying that we definitely WILL, but designing that in will make it
potentially a LOT more difficult if we do.

Remulos, 2015-10-15:
I have to agree, I like wireless in everything as much as the next guy but working for UL means I know how much of a complete pain in the arse it can be and I don't think it's worth it. SD card would work though.

dhprice, 2015-10-15:
Remind me why do we need to get compliance approval for an open source 
design?

Remulos, 2015-10-15:
If you intend to sell it you need compliance approval, regardless of the copyright or lack thereof. Licencing is not Regulation.

robot-army, 2015-10-15:
Because I'd like a product that can be 'put in to service'.

In my opinion, one big thing that's missing from open-source hardware is
standards compliance.

It can be found in software (eg. POSIX), why not hardware?

dhprice, 2015-10-15:
I fully understand that aspect, it was not made clear that the goal is 
to build a sellable product.

robot-army, 2015-10-15:
To clarify "If you intend to sell it you need compliance approval" - Not
only if you intend to sell it, but if you intend to put it in to service. I
think that's the wording, there's not many exemptions.

robot-army, 2015-10-15:
Here's a quote from (a biased) interpretation the machinery directive, not
sure if it applies to the applicable directives for this product but...

"The Directive and the European guidance makes it very clear that Machinery
built for own use must be CE marked"

http://www.cemarkingassociation.co.uk/machinery-built-for-own-use-ce-marking/

robot-army, 2015-10-15:
Can we move this thread of discussion over to the compliance requirement
issue please?

jeremypoulter, 2015-10-15:
There are versions of the ESP8266 modules that come pre-certified, but that
being said I think maybe an SD card would be more appropriate for this task.

Remulos, 2015-10-15:
The end product would still need to be tested to the EMC standard EN 301 489-1

dhprice, 2015-11-04:
V. nice to have.

robot-army, 2015-11-05:
Hardware requirement is provide some removable data storage. Is this one accepted? I can't remember.

jeremypoulter, 2015-11-05:
I think it was. This would also be really useful in the dev stage for
debugging as well.

robot-army, 2015-11-05:
OK

closed 2015-11-05 by robot-army